xpack-dev-tools/openocd-xpack #2: OpenOCD fails with Atmel ICE probe

Binuton, 2020-07-06:
### Description

OpenOCD fails to launch with an Atmel ICE probe and custom board equipped with a SAMD51G18.
Flash, erase and debug session can be launched when using MPLABX or ATMEL STUDIO with the Atmel probe.
I am able to launch debug session with gdb and OpenOCD when using an STM32F4-Disco (STM32F407 and Stlink).
I have tried to reduce speed of the adapter from its initial 2000 kHz to 500kHz/50kHz/20kHz and even 2kHz.

### Steps to Reproduce

`openocd -d3 -f myboard_samd51.cfg`

Content of "myboard_samd51.cfg":
`source [find interface/cmsis-dap.cfg]`
`transport select swd`
`cmsis_dap_vid_pid 0x03eb 0x2141`
`set CHIPNAME samd51g18a`
`source [find target/atsame5x.cfg]`
`reset_config srst_only`

**Expected behaviour:**

I expect to be able to launch a command line gdb (arm-none-eabi) session using "target extended-remote localhost:3333"

**Actual behaviour:** 

I end up with following extracted message from OpenOCD when using debug level 3:

`Debug: 214 506 cmsis_dap_usb.c:730 cmsis_dap_swd_read_process(): SWD ack not OK @ 0 JUNK`
`Debug: 215 511 command.c:626 run_command(): Command 'dap init' failed with error code -4`
`User : 216 513 command.c:692 command_run_line():`
`Debug: 217 515 command.c:626 run_command(): Command 'init' failed with error code -4`
`User : 218 517 command.c:692 command_run_line():`
`Debug: 219 519 target.c:1978 target_free_all_working_areas_restore(): freeing all working areas`
`Debug: 220 523 cmsis_dap_usb.c:932 cmsis_dap_swd_switch_seq(): SWD-to-JTAG`

### Versions

- OpenOCD version: 0.10.0-14
- operating system: Windows 10

ilg-ul, 2020-07-06:
Thank you for your bug report, but please enter it via the OpenOCD support channels, I am only packing the upstream sources.

This tracker is only for issues related to packing, not related to functionality.

closed 2020-07-06 by ilg-ul

ilg-ul, 2020-07-06:
@Binuton, just curious, when you entered the bug, didn't you get a message redirecting functional issues to the OpenOCD support channels?

Binuton, 2020-07-06:
@ilg-ul , Sorry ! I got confused because in these support channels, the OpenOCD's dedicated section of the Sparkfun forum is closed.

ilg-ul, 2020-07-06:
I don't know about Sparkfun, but the text I used for the issues gives a link to OpenOCD discussions; isn't that link useful?

And anyway, I thought that the text is clear enough that here are expected only issues related to building packing OpenOCD, not related to functionality. Isn't that text clear enough?

If the text is not clear, please suggest improvements.