xpack-dev-tools/openocd-xpack #8: How to publish link

CInsights, 2021-03-15:
Hi Liviu,
How to publish link on  from https://xpack.github.io/openocd/ points to https://github.com/xpack-dev-tools/openocd-xpack/blob/xpack/README-PUBLISH.md
It should now link to https://github.com/xpack-dev-tools/openocd-xpack/blob/xpack/README-RELEASE.md
--
Bob

ilg-ul, 2021-03-15:
Hi Bob, thank you for the report, I fixed it.

For future issues related to the web site, please use the separate tracker: 

- https://github.com/xpack/web-jekyll/issues.

closed 2021-03-15 by ilg-ul

CInsights, 2021-03-15:
Hi Liviu,
Ok on correct reporting location. Rushed that one out last night and didn’t find the link.

BTW the xpack.GitHub.io/openocd/ pages now have a visible edit button at the bottom when I view them in Edge. Just 404s but should not be there I expect.
—
Bob

(Sent from iPhone)

  *   https://github.com/xpack/web-jekyll/issues.

ilg-ul, 2021-03-16:
>  the xpack.GitHub.io/openocd/ pages now have a visible edit button at the bottom ... 404s 

A broken link; fixed, thank you.

BTW, please mind the way you write messages, since markdown will be confused by various characters and generate weird formatting. (I had to edit your initial post since it was hard to read).

Also prefer the web interface, since you have a better control. If you insist on replying via email, be sure you remove the original message, otherwise you get posts like the one above.