xpack-dev-tools/openocd-xpack #9: stm32f2x.cfg missing reset part which causes problem - you cannot debug Nucleo-F207ZG

MariwanJ, 2021-04-28:
### Disclaimer

The xPack OpenOCD is only a binary distribution of the standard source code
OpenOCD, and does not intend to add new functionality, or to fix existing
problems.

If you have questions/suggestions related to the procedure used to
build/publish the binary distribution, continue reading the next sections.

If you have problems related to OpenOCD functionality, please use the
official [OpenOCD support channels](http://openocd.org/discussion/).

### Prerequisites

Before entering a new ticket, please consider the following:

- [ ] read the documentation pages carefully (`share/doc/pdf/openocd.pdf`)
- [ ] check the [xPack Forums](https://www.tapatalk.com/groups/xpack/) for similar problems
- [ ] check the [GitHub Issues](https://github.com/xpack-dev-tools/openocd-xpack/issues/)

If you still could not find a solution, if you have interesting use
cases, if you have custom configurations, and generally if you have
any experience that you want to share with others, use the
[xPack Forums](https://www.tapatalk.com/groups/xpack/), the OpenOCD section.

If you are convinced that you identified a bug related to the binary
distribution, not OpenOCD itself (if you have doubts, use the forum),
or you have a pertinent suggestion how to enhance the xPack OpenOCD
distribution, continue and register a new issue.

### Description
The scritp is missing reset part which causes a big problem. You cannot debug Nucleo-F207ZG
[Description of the bug or feature]

### Steps to Reproduce
Trying to debug 
1. Trying to debug Nucleo-F207ZG board with any mbed project exporting it to GNU-Make. and using eclipse to debug with opeoncd
**Expected behaviour:** [What you expected to happen]
debug is failing .. cannot reset - halt the MCU
**Actual behaviour:** [What actually happened]

### Versions
Latest
- [OpenOCD version]
- [operating system]

Please understand that without being able to reproduce the bug we cannot
identify your problem.
#https://github.com/ntfreak/openocd/pull/12
---

Note: Remove unnecessary text after reading, before entering your ticket.

ilg-ul, 2021-04-28:
Sorry, but did you read the text you just posted?

closed 2021-04-28 by ilg-ul

MariwanJ, 2021-04-28:
Yes, but they didn't read the pull request. I know it should be PR in another github but I cannot find it. 
I know you just distribute it .. but you might be able to reach them? Don't know.
Sorry if the issue creation is not so good.. Don't have much experience with these kind of things. But I wanted to share with you a work of several days. It was hard to find the issue.

ilg-ul, 2021-04-28:
http://openocd.org/discussion/mailing-lists/