xpack-dev-tools/openocd-xpack #16: jtagspi driver with support for 4-byte addresses (needed for large Flash Memories)

ilg-ul, 2021-11-22:
### Discussed in https://github.com/xpack-dev-tools/openocd-xpack/discussions/15

<div type='discussions-op-text'>

<sup>Originally posted by **SteffenKlu** November 22, 2021</sup>
The jtagspi driver was updated this month (https://review.openocd.org/c/openocd/+/4876/) and I am interested if
you could build a new 'xpack-openocd-0.11.xx' Tar-ball which includes those latest improvements?
I tried to flash the large SPI-Flash in an Xilinx-Board, with the existing openocd release but it failed:

[Log]
xPack OpenOCD x86_64 Open On-Chip Debugger 0.11.0+dev (2021-10-16-21:15)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
xc7_program
jtagspi_program
Info : usb blaster interface using libftdi
Error: unable to get latency timer
Info : This adapter doesn't support configurable speed
Info : JTAG tap: xc7.tap tap/device found: 0x13631093 (mfg: 0x049 (Xilinx), part: 0x3631, ver: 0x1)
Info : JTAG tap: xc7.tap tap/device found: 0x13631093 (mfg: 0x049 (Xilinx), part: 0x3631, ver: 0x1)
Info : Found flash device 'sp s25fl256s' (ID 0x00190201)
Warn : device needs paging or 4-byte addresses - not implemented
[...]
Info : sector 9 took 2 ms
Info : Found flash device 'sp s25fl256s' (ID 0x00190201)
Warn : device needs paging or 4-byte addresses - not implemented
read 645496 bytes from file build/results/ip0170p.ppn.bin and flash bank 0 at offset 0x00000000 in 12.880716s (48.939 KiB/s)
contents differ
diff 0 address 0x00000000. Was 0x00 instead of 0xff
diff 1 address 0x00000001. Was 0x09 instead of 0xff
diff 2 address 0x00000002. Was 0x0f instead of 0xff
diff 3 address 0x00000003. Was 0xf0 instead of 0xff
diff 4 address 0x00000004. Was 0x0f instead of 0xff
[...]
More than 128 errors, the rest are not printed.
[/Log]

I hope the new enhancements to the jtagspi driver solve this problem.

Best,
Steffen</div>

SteffenKlu, 2021-11-24:
A test with a self compiled openocd version shows that the addressed topic is indeed fixed on the master branch of the openocd git repository:

[Log]
Open On-Chip Debugger 0.11.0+dev-00501-g4626af440 (2021-11-24-07:40)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
xc7_program
jtagspi_program
Info : usb blaster interface using libftdi
Error: unable to get latency timer
Info : This adapter doesn't support configurable speed
Info : JTAG tap: xc7.tap tap/device found: 0x13631093 (mfg: 0x049 (Xilinx), part: 0x3631, ver: 0x1)
Info : JTAG tap: xc7.tap tap/device found: 0x13631093 (mfg: 0x049 (Xilinx), part: 0x3631, ver: 0x1)
Info : Found flash device 'sp s25fl256s' (ID 0x190201)
Warn : 4-byte addresses needed, might need extra command to enable
flash 'jtagspi' found at 0x00000000

Info : sector 0 took 1708 ms
Info : sector 1 took 1728 ms
Info : sector 2 took 95 ms
Info : sector 3 took 100 ms
Info : sector 4 took 99 ms
Info : sector 5 took 97 ms
Info : sector 6 took 95 ms
Info : sector 7 took 95 ms
Info : sector 8 took 98 ms
Info : sector 9 took 96 ms
read 645496 bytes from file build/results/ip0170p.ppn.bin and flash bank 0 at offset 0x00000000 in 12.291559s (51.285 KiB/s)
contents match
[/Log]

ilg-ul, 2021-11-24:
Ok, thank you for confirming, then it'll be included in the next release.

ilg-ul, 2021-12-07:
Please check the latest v0.11.0-3 release and let me know if it solves this problem.

SteffenKlu, 2021-12-08:
> Android [4].
> Links:

Hi Liviu,

yes, the topic is solved with release v0.11.0-3.
Thank you for the prompt help.

Best,
Steffen

ilg-ul, 2021-12-08:
Ok, thank you for confirming.

And please don't reply via email, use the web interface. Or at least remove the useless quoted message.

closed 2021-12-08 by ilg-ul