xpack-dev-tools/openocd-xpack #33: 0.12.0-4 seems to be missing libusb?

PerennialNovice, 2024-12-12:
I use openocd integrated in STM32duino 2.9.0 and I get the following error message:

```
xPack Open On-Chip Debugger 0.12.0+dev-01685-gb9224c0c0-dirty (2024-08-02-19:51)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'gdb port', not 'gdb_port'
DEPRECATED! use 'tcl port' not 'tcl_port'
CDRTOSConfigure
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Info : clock speed 500 kHz
Error: libusb_open() failed with LIBUSB_ERROR_NOT_FOUND
Info : STLINK V2J45M31 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.239765
Error: init mode failed (unable to connect to the target)

[2024-12-12T11:35:42.381Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

[according to STM32duino ](https://github.com/orgs/stm32duino/discussions/2603) this is an xpack openocd issue and could be remedied by installing libusb on the machine

fpistm, 2024-12-12:
Issue not present with previous version.

ilg-ul, 2024-12-12:
What is your platform?

ilg-ul, 2024-12-12:
I cannot confirm this:

```console
% tree -L 2 xpack-openocd-0.12.0-[34]
xpack-openocd-0.12.0-3
├── README.md
├── bin
│   ├── libftdi1.dll
│   ├── libusb-1.0.dll
│   └── openocd.exe
├── distro-info
│   └── licenses
└── openocd
    ├── OpenULINK
    ├── angie
    ├── contrib
    └── scripts
xpack-openocd-0.12.0-4
├── README.md
├── bin
│   ├── libftdi1.dll
│   ├── libusb-1.0.dll
│   └── openocd.exe
├── distro-info
│   └── licenses
└── openocd
    ├── OpenULINK
    ├── angie
    ├── contrib
    └── scripts

18 directories, 8 files
 % ls -lL xpack-openocd-0.12.0-[34]/bin/libusb-1.0.dll
-rwxr-xr-x@ 1 ilg  staff  223695 Apr  2  2024 xpack-openocd-0.12.0-3/bin/libusb-1.0.dll
-rwxr-xr-x@ 1 ilg  staff  223695 Aug  2 22:51 xpack-openocd-0.12.0-4/bin/libusb-1.0.dll
% cmp -b xpack-openocd-0.12.0-3/bin/libusb-1.0.dll xpack-openocd-0.12.0-4/bin/libusb-1.0.dll
xpack-openocd-0.12.0-3/bin/libusb-1.0.dll xpack-openocd-0.12.0-4/bin/libusb-1.0.dll differ: char 137, line 2 is 201 ? 327 ?
```

What else might it be?

PerennialNovice, 2024-12-12:
well, I got rid of the error after downloading zadiq 2.9 and assigning libusb to the ST-Link debug probe
downside of this action: i cannot reproduce it

shall I try to undo?

ilg-ul, 2024-12-12:
My experience with Windows is minimal, I'm afraid I cannot help you.

But I would say that both versions should behave the same.

PerennialNovice, 2024-12-12:
well, it seems to be a "one of" thing and with a little help from @fpistm I managed to solve it
So I guess this one can be closed

closed 2024-12-12 by ilg-ul

fpistm, 2024-12-12:
Honestly, don't know. One weird things is it seems it only fails for a dedicated target. I've asked to OP to update the STLink firmware of the board.

ilg-ul, 2024-12-12:
Ok.

At any point, if you have suggestions to improve the documentation, please let me know.