xpack-dev-tools/openocd-xpack #34: Unable to install latest windows package

fpistm, 2025-02-06:
Hi @ilg-ul 
First thanks for the new release.
Starting to integrate it to stm32duino and met an issue only on Windows when Arduino IDE 2 install it:

```
Installing STMicroelectronics:xpack-openocd@0.12.0-5
Failed to install platform: 'STMicroelectronics:stm32:2.10.0-dev'.
Error: 13 INTERNAL: Cannot install tool STMicroelectronics:xpack-openocd@0.12.0-5: extracting archive: Create file C:\Users\<username>\AppData\Local\Arduino15\tmp\package-3714540719\xpack-openocd-0.12.0-5\openocd\angie  \angie_bitstream.bit: open C:\Users\<username>\AppData\Local\Arduino15\tmp\package-3714540719\xpack-openocd-0.12.0-5\openocd\angie  \angie_bitstream.bit: The system cannot find the path specified.
```

I've tried to manually extract on windows and it works. 
Using the arduino-cli to install it produce the same issue.
Installing the previous one works (0.12.0-4).

Is there any change between 0.12.0-4 and 0.12.0-5 around the zip creation?

ilg-ul, 2025-02-06:
> Is there any change between 0.12.0-4 and 0.12.0-5 around the zip creation?

No. 

But I also noticed that one of the names includes some spaces.

I'll investigate where they come from.

ilg-ul, 2025-02-07:
Fixed in 0.12.0-6, on 2025-02-07.

closed 2025-02-07 by ilg-ul

fpistm, 2025-02-07:
Thanks again @ilg-ul 
It fixes the issue.

ilg-ul, 2025-02-07:
You're welcome!

Did you also update Arduino to use the latest toolchain?

fpistm, 2025-02-07:
Yes in a dev  branch. This is how I see the issue on Windows. On Linux no issue even with spaces.
This will be available on main with the next release planned end of February including latest gcc.

ilg-ul, 2025-02-07:
Thank you for confirming.

FYI, although I guess it'll not affect you, the next release of the build scripts will require Linux GLIBC 2.28 (Debian 10 and up), macOS 11.0 and up, will no longer include Arm 32-bit images and probably will include Windows arm64 binaries.